citrineos/citrineos #15: Docker image under windows

goRaspy, 2024-03-24:
**Is your feature request related to a problem? Please describe.**
Quick Start guide refer to linux .sh files

**Describe the solution you'd like**

Since my objective is to run the docker image. It should probably be possible to works under windows 

Many thanks for your advice

thanaParis, 2024-03-25:
Hi @goRaspy !
If you want to run the docker image, you can skip the unix-init-install-all.sh file and simply run `docker compose up` from Server/ and it will work.
If you want to run locally, the .sh script can be executed from Git Bash, a unix-like command line utility that usually comes installed on windows machines (or, you can download it).

goRaspy, 2024-03-25:
Thank you very much for your help. Looks like it run well. Unfortunately I do not have access to anything on 127.0.0.1:8055 is there something else to do to access the web interface ?

closed 2024-03-26 by thanaParis