citrineos/citrineos #17: Documentation not working

goRaspy, 2024-03-29:
Hello,

Following the quick start guide, the documentation should be available at http://localhost:8081/docs

Unfortunately, it does not work for me : 
[image]

At short terms could you precise me the default endpoint url (ws) to setup on charging station side (something like ws://127.0.0.1:8080/websocket/CentralSystemService/ChargePointID) ?  Many thanks

goRaspy, 2024-03-29:
http://localhost:8080/docs/ works fine !  Sorry for that.

I am still looking for chargepoint endoint :(

thanaParis, 2024-03-29:
@goRaspy `ws://localhost:8081` should work after using `docker compose up`

More broadly speaking, this is controlled by [a config file](https://github.com/citrineos/citrineos-core/blob/8c3c1b36d3bb5793d6886428910ac0e0f430ec58/Server/src/config/envs/docker.ts#L77). You can change the ports used or the security profiles (security profile 0 is unsecured, easiest for testing).

goRaspy, 2024-03-29:
Thank you very much.

Will test asap

ChrisWeissmann, 2024-05-22:
Hi @goRaspy, did this get resolved? If yes, hoping to close this issue  😃

closed 2024-07-11 by thanaParis