citrineos/citrineos #20: Starting with Docker Compose as described in README fails on Apple Silicon (ARM-based) Macbook

volkert-fastned, 2024-04-24:
**Describe the bug**

When starting CitrineOS with Docker Compose on a local Apple Silicon Mac(Book), `ocpp-db` fails to start, apparently because no ARM64/Aarch64 variant of that Docker container image is available.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone https://github.com/citrineos/citrineos-core`
2. `cd citrineos-core/Server`
3. `docker-compose -f ./docker-compose.yml up -d`
4. See error

**Expected behavior**
CitrineOS should start successfully in the local environment, as described in the README.

**Screenshots**

Instead of a screenshot, pasting the error message below:

```text
! ocpp-db The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
```

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.4.1
 - client architecture: Apple Silicon (ARM64/aarch64)

thanaParis, 2024-04-24:
Thanks for reporting this!

[It looks like this is an open issue with the postgis docker image](https://github.com/postgis/docker-postgis/issues/216) 

I will look into solutions from that thread that we may be able to work into CitrineOS.

fabiolnm, 2024-04-28:
@thanaParis @volkert-fastned although this warning indeed exists, it doesn't caused a failure. 
However, the server-directus container is not starting correctly, because healthcheck fails - this can be observed by inspecting the container:

[image]

```
        "Health": {
            "Status": "unhealthy",
            "FailingStreak": 4,
            "Log": [
                {
                    "Start": "2024-04-28T04:00:27.514599215Z",
                    "End": "2024-04-28T04:00:27.548334423Z",
                    "ExitCode": 1,
                    "Output": "Connecting to localhost:8055 ([::1]:8055)\nwget: can't connect to remote host: Connection refused\n"
                },
```

I found this workaround - replaced `localhost` by `127.0.0.1` at this line:

https://github.com/citrineos/citrineos-core/blob/f7483cc3ed9751b957e34d0befa7608db391dd5d/Server/docker-compose.yml#L139

volkert-fastned, 2024-04-29:
With the merging of #81, I can confirm that the error no longer occurs and that CitrineOS starts up with Docker Compose on Mac hardware with an Apple Silicon CPU without errors now. 👍 Thanks!

closed 2024-04-29 by volkert-fastned