citrineos/citrineos #21: Official Helm chart and Helm Repo for deploying CitrineOS to Kubernetes clusters

volkert-fastned, 2024-04-25:
**Is your feature request related to a problem? Please describe.**
Many organizations use some variant of Kubernetes (AKS, EKS, GCP, etc) for managing their deployed containers. CitrineOS comes with a Docker Compose file out of the box, which eases local deployments. But for convenient deployment to a Kubernetes cluster, something like a Helm chart would be more suitable.

**Describe the solution you'd like**
It would be great if there was an official CitrineOS Helm repository, through which CitrineOS could be installed and deployed in a Kubernetes cluster in a simple and standard fashion, like this:

```shell
helm repo add citrineos https://citrineos.github.io/charts
helm repo update
helm install -f custom-config-values.yaml citrineos citrineos/citrineos
```

Note that the URL for the official Helm repo could be something different. The URL I mentioned in the example commands is just a suggestion. But it should be a URL that makes it clear that it's an officially Helm repo "blessed" by S44 and/or the official CitrineOS project maintainers.

Such a CitrineOS Helm repo should probably also be listed in the [Artifact Hub](https://artifacthub.io/). Right now, searches for "citrineos" or "CitrineOS" don't yield any results.

**Describe alternatives you've considered**
There is a tool called [Kompose](https://kompose.io/user-guide/#alternative-conversions) that can convert a Docker Compose yaml file to a Helm chart, but an official (tested) Helm chart in an official Helm repo for CitrineOS would be a lot more convenient.

volkert-fastned, 2024-04-25:
Also, I recommend that the official Helm chart should configure `Release.Namespace` to install CitrineOS to a Kubernetes namespace called `citrineos` by default. (As far as I know, Kubernetes namespaces are always in all lower case.)

ChrisWeissmann, 2024-05-22:
@volkert-fastned Thanks for the input! I think that's a great point. We are heads down on some features, I think this moves to the top of the list right after the 1.3 release. We have now started automatically publishing docker images, so this will mesh nicely.

volkert-fastned, 2024-06-05:
Sounds good! When you start working on this, you might want to take a look at a tool called [Kompose](https://kompose.io/) that can automatically convert Docker Compose yaml files to Helm charts. You may have to manually tweak a few things here and there in the generated chart afterwards, but it could be a good starting point.

If you're looking for an example on how to use it, you could take a look at the reproduction script I shared in [this EVerest issue I reported](https://github.com/EVerest/everest-demo/issues/54#issue-2336107682).

I hope that helps!

javdrher, 2024-12-27:
just a heads up, we are currently building a helm chart internally to deploy citrineos to kubernetes. We may contribute this in the near future.

jyothiram266, 2025-02-26:
Hello guys just to mention it here i have made a working helm chart for citrine os stack have a look at here  https://github.com/jyothiram266/citirin-k8s-depolyment