citrineos/citrineos #22: npm Error when installing citrineOS

tefelberson, 2024-05-05:
Hi everyone!
i'm installing Citrineos , i want to connect it to everest(https://everest.github.io/) but I'm encountering an error when executing "docker-compose up -d" command.
You can capture this screen. 
[image]

I need help please

thanaParis, 2024-05-06:
Hi @tefelberson !
Here is a stackoverflow page that elaborates on the error you are seeing: https://stackoverflow.com/questions/63010779/npm-err-code-eai-again-error-when-trying-to-install-express
I would suggest trying some of the fixes mentioned there.

ChrisWeissmann, 2024-05-22:
Hi @tefelberson, did Thana's suggestion resolve the issue for you? Seems it might be due npm running through a proxy and not being able to resolve an address. 
If all is well, can we close this issue?

closed 2024-07-11 by thanaParis