citrineos/citrineos #23: Connect charging station to CitrineOs

tefelberson, 2024-06-16:
Hello everyone,

I am starting my research project on the cybersecurity of electric vehicle charging stations. I have taken the time to study and understand how the OCPP protocol works, and I need a simulation lab. **Citrineos** is one of the most popular (if not the most popular) open-source CSMS. I would like to connect charging stations to Citrineos. I have tried **Everest** and **MicroOCPP**, but I have not yet been able to successfully communicate with **Citrineos** using these tools. Can you help me? Could you provide me with the steps to follow?

thanaParis, 2024-06-17:
Hi @tefelberson !
We have set up a demo with EVerest: https://github.com/EVerest/everest-demo/pull/58
You can try pulling [the branch](https://github.com/ChrisWeissmann/everest-demo/tree/feature/add-citrine-support-signedoff) from that pr and running the demo.

Overall, here are the steps:

1. Clone CitrineOS, run docker compose up from Server/
2. Review the websocket servers in [CitrineOS's system configuration](https://github.com/citrineos/citrineos-core/blob/51f665f80314fff39f5e03eb226ee3a9471c8670/Server/src/config/envs/docker.ts#L79). There is one port per security profile, and an unsecured (security profile '0') port. Running locally from the main branch, CitrineOS will have websocket servers at ws://localhost:8081, ws://localhost:8082, wss://localhost:8443, and ws://localhost:8444. I would suggest using the unsecured port (8081) to begin with.
3. Set your charger to connect to the chosen ws url, say ws://localhost:8081. For EVerest, this means making sure the NetworkConfigurationProfiles device model Variable has an Actual Variable_Attribute using this url, keeping in mind that EVerest will also expect the stationId to already be appended (i.e., `{"configurationSlot": 1, "connectionData": {"messageTimeout": 30, "ocppCsmsUrl": "ws://localhost:8081/cp001", "ocppInterface": "Wired0", "ocppTransport": "JSON", "ocppVersion": "OCPP20", "securityProfile": 1}}`)

tefelberson, 2024-06-20:
Thank you for your reply!
I tried setting up the Everest demo script several times, but ran into a problem.
An error occurred while creating the server-directus container.

[image]

tefelberson, 2024-07-06:
someone has it please.
@thanaParis 

do you have a solution for me?

thanaParis, 2024-07-11:
Do you have any logs from the Directus container? Did you try restarting it? How long did you wait for the demo to come online? Sometimes it takes a minute for CitrineOS to start up; did you review the citrine container's logs to see if it was having issues when starting?

I will need some additional info to get to the bottom of your issue. Please provide some logs and try restarting the containers if they crash at first.

yuvenus, 2024-09-04:
Hello @tefelberson! Following [release 1.3.1](https://github.com/citrineos/citrineos-core/releases/tag/v1.3.1), we've added an easier way to get EVerest and CitrineOS running together via the `npm run start-everest` command.

This command will spin up an EVerest instance that should communicate with your local CitrineOS instance as long as it is running. You can read more about it in the README found in `./Server/everest`.

If you have a chance, can you please pull the latest release of CitrineOS on `main`, refresh your NPM caches with the command `npm run fresh-and-install-all`, and try out the new aforementioned EVerest script? Thank you!

und3rd06012, 2025-06-29:
Hello! 

I am trying to do the same thing and I am using release 1.6.3. I have also been able to use and run directus.
I am following this guide here: https://github.com/citrineos/citrineos-core/blob/main/Server/everest/README.md

When I am trying the first method (non manual)

I execute this:

sudo env "PATH=$PATH:/home/my_username/.nvm/versions/node/v24.3.0/bin" npm run start-everest

EVerest starts but I get also this on my terminal:

[out2.txt](https://github.com/user-attachments/files/20970272/out2.txt)

The user interface on 127.0.0.1:1880/ui looks like this:

[image]

and basically when I am trying to start the car charging nothing happens. (the meter shows nothing)

When I am trying the manual method I am stumble upon the same issue as @tefelberson with HTTP 500 errors

CitrineOS and Directus seem to run ok. I can see all that the database has been bootstrapped in directus.

I am sorry if my question is trivial. I am not very familiar to all this.

I am also trying to do some security research.

Thanks