citrineos/citrineos #24: Connect IoCharger to CitrineOS

DylanVautrin, 2024-07-02:
Hello ! 
We are working with IoCharger charger and we would like to connect it to citrine os.

We succeed to deploy on a distant server, but we never succeed to connect a charger to it.
Should we use port 8081 or 8082 ? And we have to open all the port that is used by CitrineOS ? 

Thanks in advance if you can help us to connect the charger, the manufacturer said that the config of the charger is fine so maybe we miss something on setup.
Best regards !

thanaParis, 2024-07-02:
Hi Dylan! 8081 should work, it is unsecured. What errors are you seeing on CitrineOS's side, if any?
Since 8081 is unsecured, the charger cannot use TLS, so make sure the connection url is `ws` not `wss`. Also, make sure that the charger is using the correct subprotocol header (ocpp2.0.1)--if the charger is capable of OCPP 1.6J and OCPP 2.0.1, it needs to be using OCPP 2.0.1. If it's the subprotocol error though, you should see logs on CitrineOS that say so (log starts with "Protocol mismatch").

If there's any more information you can provide, I can offer more advice.

DylanVautrin, 2024-07-02:
Hi !
Thanks for your help.
Where i can check Citrine logs ? 
Also, i have another issue, my directus is "unhealthy" but i can access it 

[image]

Each time i try to start it again it have an issue, and last time it started but unhealthy, not really sure of what i missed in setup config.

Thanks again for your help !

DylanVautrin, 2024-07-03:
Hello again, 
I still have some issues but i succed to get the log during the start using the command : docker compose up --force-recreate.
Here the log : 
[Log - start citrineos.txt](https://github.com/user-attachments/files/16086558/Log.-.start.citrineos.txt)

Thanks for your help !
Best regards

thanaParis, 2024-07-03:
[I've found a thread which seems to explain your issue](https://github.com/directus/directus/discussions/6480).
Please try some of the solutions at that link!

DylanVautrin, 2024-07-04:
Hello 
I think i've succeed to deploy it correctly, but i only got the log from directus, where in can find the ocpp log please ? 
thanks

DylanVautrin, 2024-07-05:
Hello ! 
I think i finally succeed to have something stable, can you just confirm me that is normal to have only 4 docker running please ? 
[image]
Thanks in advance !

thanaParis, 2024-07-05:
Hi Dylan! Yes it is normal to only have 4 docker containers running. There is a swarm mode which separates citrine into microservices for horizontal scaling, but the normal mode runs all the modules on the same service.

DylanVautrin, 2024-07-05:
OKay thanks ! Where i can check the OCPP log please ? I xant to add a RFID token to CitrineOS how can i do ? 
Thanks

thanaParis, 2024-07-05:
> Where i can check the OCPP log please ? 

There's a few different ways. If you hook into the citrine docker container's shell, you can see citrine's logs and that will include the ocpp messages.

You can also use the subscription API, available at port 8080, path /data/ocpprouter/subscription
With a POST and the following JSON body:
{
    "stationId": "string",
    "onConnect": bool,
    "onClose": bool,
    "onMessage": bool,
    "sentMessage": bool,
    "url": "https://url.to.post.endpoint"
}
The url you pass in will have the chosen events for the given station POSTed to it.

> I xant to add a RFID token to CitrineOS how can i do ?

Take a look at localhost:8080/docs when running CitrineOS to see the full OpenAPI spec. The subscription api from above is there, as are some CRUD endpoints for adding data entities such as Authorize.

You'll want this path `/data/evdriver/authorization?idToken=<string>&type=<string>`
PUT JSON body:
{
  "idToken": {
    "idToken": "<string>",
    "type": "NoAuthorization" <- for RFID, this is usually ISO14443, depending on the ISO standard of the rfid chip you are looking to add.
  },
  "idTokenInfo": {
    "status": "Accepted"
  }
}

DylanVautrin, 2024-07-09:
Hello, 
Thanks again for your help, I found for the log.
When adding a token, if i'm doing it right (it got response 200 so i think yes), it should appear in directus GUI no ? 
Thanks

DylanVautrin, 2024-07-09:
Hello, 
I think i succeed to add a token, but i don't understand how to delete it, on GUI i got an error when i try, and the API request ask for a stationID , if i try with my token ID it don't work.
Can i have a hint please ? 
Thanks :)

DylanVautrin, 2024-07-12:
For now, it should be good ! 
Thanks again for your help :)

closed 2024-07-12 by DylanVautrin