citrineos/citrineos #25: Issues with "Components Variable"

DylanVautrin, 2024-07-09:
Hello,
I have an issue with my CitrineOS running, everything is fine but i don't have access to "Componenents Varialble" module : 
[image]

Did i make a mistake during the build ? 
Thanks !

yuvenus, 2024-07-11:
Hello Dylan, you didn't make any mistake in your build, don't worry! The Directus UI does not properly display junction tables such as `ComponentVariable`, because there is no ID in the table, which is caused by Sequelize creating compound index off the two existing columns, `componentId` and `variableId`, rather than using an ID as the primary key. 

Unfortunately, we can't fix the Directus display as a result 😔 If you need to view the contents of this table, the best way would be through a database GUI. In general, you shouldn't need to directly add items to this table, but you can do so through a database GUI in case of emergency.

Please let us know if you have any more questions, and if not, please let us know if this issue can be closed. Thank you!

DylanVautrin, 2024-07-12:
Oh okay, then i close this issue if it is okay ! 
Thanks :)

closed 2024-07-12 by DylanVautrin