citrineos/citrineos #26: Using "GetVariables" command

DylanVautrin, 2024-07-12:
Hello,
I'm trying to use the command “GetVariable”, I succeed to get one existing parameter, with this body : 
` {
  "getVariableData": [
    {
      "component": {
        "name": "ChargingStation"
      },
      "variable": {
        "name": "SerialNumber"
      }
    }
  ]
} `

But I would like to make the request and get ALL parameter, how can I do it please ? 
Btw, if I make a mistake in the body, like putting “string” for the name like the example, it “create” a parameter in the GUI, and I don't understand why, because it is GetVariable and not set aha, 
Thanks for your help :)

thanaParis, 2024-07-15:
GetVariables is still a source of information from the charger. It doesn't set the variable on the charger, but it can be used to make CitrineOS aware of not-previously-known variables, which is why it creates a parameter in the GUI.

If you want to load in the entire device model, you can do that with GetBaseReport of type 'FullInventory'. There is an option in the Boots table to do this automatically when the boot is Pending.

DylanVautrin, 2024-07-16:
Hello, 
Thanks for your help, looks like what I was looking for :) 
So, the "GetVariable" is more likely to get value of one or multiple particular parameter of a charger ?

thanaParis, 2024-07-16:
Exactly. The GetVariables message is used to retrieve specific variable values, but is impractical for loading the entire device model.

If this answers your question, please close this thread

DylanVautrin, 2024-07-17:
Okay thanks ! 
I'm closing this :)

closed 2024-07-17 by DylanVautrin