citrineos/citrineos #27: "SetVariable" on ReadOnly modify the parameter

DylanVautrin, 2024-07-12:
Hello, 
I was trying to set some parameter of my charger, and I tried one in Read Only to try. The charger response was rejected, but the parameter in the “Variable Attribute” windows have been changed, despite the response was rejected and parameter set to “Read-only”.
Is that a bug ? Or I made something wrong ?
Thanks for your help.

thanaParis, 2024-07-15:
Hi @DylanVautrin ! That could be a bug. There's a collection, VariableStatus, that is supposed to capture the current state of the variable attributes. When a variable fails to set, there should be a record of it in the VariableStatuses, then the VariableAttribute should be rolled back to the last successful value. The only exception to this is when the device model is updated with the 'setOnCharger' query param as true, indicating that charger already has this configuration and CitrineOS doesn't need to confirm.

If you could confirm what actions are needed to create this issue, and show some screenshots of the collections, we can investigate this further.

DylanVautrin, 2024-07-16:
Hello, 
By luck, I made a screenshot of my test, so here the command I've sent : 
[image]

I have the response of the charger too if needed : 
[image]

And here the content of Variable Statuses : 
[image]

Do you need something else ? 

Thanks again :)

thanaParis, 2024-07-16:
This is definitely a bug, I see the issue in the code. There should be logic to roll back to the most recent accepted value in the 'updateResultByStationId' method of the SequelizeDeviceModelRepository when the result is rejected.

Thank you for pointing this out! If you feel comfortable fixing it yourself, please open a PR. Otherwise, I'll update this thread in the future when someone else gets to fixing the issue.

DylanVautrin, 2024-07-17:
Hello,
Okay ! Unfortunately, i don't think i'm able to make the fix myself 
Thanks again for your help

yuvenus, 2024-09-04:
Hello @DylanVautrin! The bugfix has been merged to [release 1.3.1](https://github.com/citrineos/citrineos-core/releases/tag/v1.3.1). 

If you have a chance, can you pull CitrineOS `main` (currently on [release 1.4.2](https://github.com/citrineos/citrineos-core/releases/tag/v1.4.2)) and see if it fixes the bug you were facing? If so, please close this issue. Thank you!

thanaParis, 2024-09-18:
Closing this as fixed.

closed 2024-09-18 by thanaParis