citrineos/citrineos #28: How to clear saved data on CitrineOS

DylanVautrin, 2024-07-12:
Hello, 
I'm trying to understand how Citrine OS works, so I made some test with different charger, and it created different item for different module, like “Variables Attributes”, or “Charging Stations”. 
How can I “reset” all, and get back a “clear” environment ? Should I delete everything one by one on GUI ?
Thanks in advance :)

thanaParis, 2024-07-15:
If you start CitrineOS with the SystemConfiguration variable data.sequelize.sync as true, it will wipe the data in all tables on start up. Does that accomplish what you are looking for?

DylanVautrin, 2024-07-16:
Hello, 
Yes sounds good, i'm going to try this, how can i set this ? When using the command docker compose up -d ? 
Thanks !

thanaParis, 2024-07-16:
You can reference this page: https://citrineos.github.io/systemconfiguration.html

DylanVautrin, 2024-07-17:
Okay thanks !  I will try it