citrineos/citrineos #29: Unable to docker compose swarm folder on windows

Prev-I, 2024-07-15:
**Which repository?**
Is it:

- citrineos-core

**Describe the bug**
I've tried to docker-compose the swarm folder for the current main branch but it won't start reporting file not found.

**Expected behavior**
A way to launch swarm compose on windows

**Desktop (please complete the following information):**
 - OS: Windows 11 with Docker desktop on WSL 2
 - Version: main (1.3.0)

Prev-I, 2024-07-17:
To integrate my previous message, the console error is the following

```
docker-compose up -d
time="2024-07-17T17:14:16+02:00" level=warning msg="C:\\Users\\previtalicl\\source\\repos\\__citrineos\\citrineos-core\\Swarm\\docker-compose.yml: `version` is obsolete"
[+] Building 0.0s (1/1) FINISHED                                                                                docker:desktop-linux
 => CANCELED [directus internal] load build definition from directus.Dockerfile                                                 0.0s
resolve : CreateFile C:\Users\previtalicl\source\repos\__citrineos\citrineos-core\Swarm\docker: The system cannot find the file specified.
```

thanaParis, 2024-07-17:
Hi @Prev-I ! This is a known issue. Unfortunately, we aren't intending on fixing Swarm for 1.3.X. Swarm support will be redesigned in a later release, likely 1.5.

Prev-I, 2024-07-17:
Thank you @thanaParis for explaination. I'll leave It on the todo-list and I'll nove forward checking other cool features of the Project!