citrineos/citrineos #30: Issues on Transactions

DylanVautrin, 2024-07-17:
Hello, 
I made some test by making some transaction. First, my module Tariff Cost is not configured, if my issue is due to this, I'm sorry :c 
I notice an issue when I was making some transaction, my test is : 
- Start a transaction with a tag on my charger, then end it the same way, so I had a new line appear on module “Transactions” : 
- 
[image]

But, when I stopped it, I noticed that the “Is Active” column was not updated. So the transaction you see with ID = "56ed3582-cb37-4da5-9b1e-ea9e5c749a7c” is noramlly stopped, a i can see in “TransactionEvent” module : 

[image]

So, we can see that I “Ended” the session : 56ed3582-cb37-4da5-9b1e-ea9e5c749a7c and started a new one, with id fcbc56c7-4ca3-4803-9975-27d8d39f91aa.

Soo, the issue is that the new transaction didn't appear in module “Transaction”, plus, the “old” one is still “Active”.
Another issue, if I try to make a “requestStopTransaction”, with one or the other transaction ID, it responds that it do not exist.

I'm not sure if I'm clear, if you need more information just ask me, and if i'm just doing something wrong i'm sorry again !
Thanks in advance for your help ! :D

thanaParis, 2024-07-17:
Hi @DylanVautrin !

> But, when I stopped it, I noticed that the “Is Active” column was not updated.

We are aware of this issue and have fixed it for 1.3.1: https://github.com/citrineos/citrineos-core/pull/166

> Soo, the issue is that the new transaction didn't appear in module “Transaction”, plus, the “old” one is still “Active”.

It's not possible in 1.3.0 for a TransactionEvent to be saved to CitrineOS without a related Transaction. Can you clarify what you're seeing here? Are the TransactionEvents associated with the wrong Transaction?

> Another issue, if I try to make a “requestStopTransaction”, with one or the other transaction ID, it responds that it do not exist.

When you say "it responds that it do not exist" could you clarify? CitrineOS has no logic for rejecting a RequestStop. Do you mean that the charger you are using said your transactions didn't exist?

DylanVautrin, 2024-07-19:
Hello, 
Thanks for your reply, 
For the fixed issues, thanks, i will try to update it.

I mean that in my TransactionsEvent, i have the end of the session "56ed3582-cb37-4da5-9b1e-ea9e5c749a7c", and the start of a new one "fcbc56c7-4ca3-4803-9975-27d8d39f91aa", but in Transactions, i only have one, the second one don't appear

And for the last issues, i'm not sure, maybe i made an error using the API, i will test it again to be sure.

thanaParis, 2024-07-24:
@DylanVautrin have you been able to test this with 1.3.1? Curious if this issue is still present; any info from new tests if it is would be really helpful.

DylanVautrin, 2024-07-25:
Hello, not really for now I'm trying to start CitrineOS with the OCPI part and I didn't succeed, that's why I opened an issue about this too :)

DylanVautrin, 2024-07-26:
Hello !
i succeed to start CitrineOS with version 1.3.1 :)
So i tried to start a Transation, with a Tag, and i got some error in the log, and the Transaction do not appear in the module : 
[image]
Do you have an idea of what happened please ? 
Thanks !

DylanVautrin, 2024-07-31:
Hello ! 
I finally tested a transaction on newest version and I got some issues : 
— First, the status of the charging stations is not updated, even few days after I stopped the charging station, it is still “Online” : 
[image]
— Then I made a Transaction, the transaction is well set to “Active” when it starts, but when i ended it no : 
[image]
[image]

 I can make other test in particular if needed,
Thanks !

thanaParis, 2024-09-18:
Closing this as fixed.