citrineos/citrineos #31: My issue name [Repository-name]

Prev-I, 2024-07-22:
**Which repository?**
Is it:

- citrineos-core

**Describe the bug**
With the latest version on citrineos-core I'm unable to build citrineos, docker build fail with npm error

**To Reproduce**
1. Checkout main branch of citrineos-core and citrineos-ocpi
2. Inside Server folder
3. docker compose build

**Expected behavior**
Build working as before (tested successfully on 1.3.0)

**Desktop (please complete the following information):**
 - OS: Windows 11, WSL2 with docker

**Output log**

```
\__citrineos\citrineos-core\Server> docker compose build
[+] Building 107.6s (26/33)
 => [directus internal] load build definition from directus.Dockerfile                                                          0.0s
 => => transferring dockerfile: 887B                                                                                            0.0s
 => [directus internal] load metadata for docker.io/directus/directus:10.10.5                                                   1.0s
 => [directus internal] load .dockerignore                                                                                      0.0s
 => => transferring context: 314B                                                                                               0.0s
 => [directus 1/6] FROM docker.io/directus/directus:10.10.5@sha256:e13f24114fb49e9198dfb6ad6f54788901406f17778332cd94b8f572795  0.0s
 => [directus internal] load build context                                                                                      0.0s
 => => transferring context: 10.04kB                                                                                            0.0s
 => CACHED [directus 2/6] COPY tsconfig.build.json /directus                                                                    0.0s
 => CACHED [directus 3/6] COPY charging-stations-bundle/tsconfig.json /directus/extensions/directus-extension-charging-station  0.0s
 => CACHED [directus 4/6] COPY charging-stations-bundle/package.json /directus/extensions/directus-extension-charging-stations  0.0s
 => CACHED [directus 5/6] COPY charging-stations-bundle/src /directus/extensions/directus-extension-charging-stations-bundle/s  0.0s
 => CACHED [directus 6/6] RUN npm install --prefix /directus/extensions/directus-extension-charging-stations-bundle && npm run  0.0s
 => [directus] exporting to image                                                                                               0.0s
 => => exporting layers                                                                                                         0.0s
 => => writing image sha256:cf370da8790aeaa169aab291ea065390c70605782b4069b27f456739402566f3                                    0.0s
 => => naming to docker.io/library/server-directus                                                                              0.0s
 => [citrine internal] load build definition from deploy.with.ocpi.Dockerfile                                                   0.0s
 => => transferring dockerfile: 1.10kB                                                                                          0.0s
 => WARN: FromAsCasing: 'as' and 'FROM' keywords' casing do not match (line 3)                                                  0.0s
 => [citrine internal] load metadata for docker.io/library/node:18-slim                                                         0.7s
 => [citrine internal] load metadata for docker.io/library/node:18                                                              0.7s
 => [citrine internal] load .dockerignore                                                                                       0.0s
 => => transferring context: 2B                                                                                                 0.0s
 => [citrine build  1/14] FROM docker.io/library/node:18@sha256:aabbaf118c7c0a6e9a3bda69bd2a94b0f6c4150bf80c501ef1c87ffcb56873  0.0s
 => [citrine stage-1 1/3] FROM docker.io/library/node:18-slim@sha256:8bc58ec22d5114a8cef0d2de719d0b88c11fa17b7a49ab6f18291ab98  0.0s
 => [citrine internal] load build context                                                                                       3.5s
 => => transferring context: 132.67MB                                                                                           3.5s
 => CACHED [citrine build  2/14] WORKDIR /usr/local/apps/citrineos-ocpi                                                         0.0s
 => CACHED [citrine build  3/14] COPY ./citrineos-ocpi .                                                                        0.0s
 => CACHED [citrine build  4/14] WORKDIR /usr/local/apps/citrineos-core                                                         0.0s
 => CACHED [citrine build  5/14] COPY ./citrineos-core .                                                                        0.0s
 => CACHED [citrine build  6/14] WORKDIR /usr/local/apps/citrineos-ocpi                                                         0.0s
 => CACHED [citrine build  7/14] RUN npm i                                                                                      0.0s
 => CACHED [citrine build  8/14] WORKDIR /usr/local/apps/citrineos-core                                                         0.0s
 => ERROR [citrine build  9/14] RUN npm i                                                                                     102.6s
------
 > [citrine build  9/14] RUN npm i:
78.76 npm warn deprecated inflight@1.0.6: This module is not supported, and leaks memory. Do not use it. Check out lru-cache if you want a good and tested way to coalesce async requests by a key value, which is much more comprehensive and powerful.
78.96 npm warn deprecated @humanwhocodes/config-array@0.11.14: Use @eslint/config-array instead
79.00 npm warn deprecated rimraf@3.0.2: Rimraf versions prior to v4 are no longer supported
79.16 npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
79.17 npm warn deprecated @humanwhocodes/object-schema@2.0.3: Use @eslint/object-schema instead
79.18 npm warn deprecated glob@8.1.0: Glob versions prior to v9 are no longer supported
79.18 npm warn deprecated glob@8.1.0: Glob versions prior to v9 are no longer supported
79.18 npm warn deprecated glob@7.2.0: Glob versions prior to v9 are no longer supported
79.72 npm warn deprecated @npmcli/move-file@1.1.2: This functionality has been moved to @npmcli/fs
80.57 npm warn deprecated are-we-there-yet@2.0.0: This package is no longer supported.
80.85 npm warn deprecated gauge@3.0.2: This package is no longer supported.
80.97 npm warn deprecated npmlog@5.0.1: This package is no longer supported.
81.34 npm warn deprecated superagent@7.1.6: Please upgrade to v9.0.0+ as we have fixed a public vulnerability with formidable dependency. Note that v9.0.0+ requires Node.js v14.18.0+. See https://github.com/ladjs/superagent/pull/1800 for insight. This project is supported and maintained by the team at Forward Email @ https://forwardemail.net
82.72 npm warn deprecated json-schema-ref-parser@6.1.0: Please switch to @apidevtools/json-schema-ref-parser
85.38 npm warn deprecated eslint-config-standard-with-typescript@38.1.0: Please use eslint-config-love, instead.
85.47 npm warn deprecated are-we-there-yet@3.0.1: This package is no longer supported.
85.51 npm warn deprecated npmlog@6.0.2: This package is no longer supported.
85.59 npm warn deprecated gauge@4.0.4: This package is no longer supported.
87.09 npm warn deprecated tslint@6.1.3: TSLint has been deprecated in favor of ESLint. Please see https://github.com/palantir/tslint/issues/4534 for more information.
101.9 npm error code ERR_INVALID_ARG_TYPE
101.9 npm error The "path" argument must be of type string or an instance of Buffer or URL. Received null
101.9
101.9 npm error A complete log of this run can be found in: /root/.npm/_logs/2024-07-22T13_43_40_012Z-debug-0.log
------
failed to solve: process "/bin/sh -c npm i" did not complete successfully: exit code: 1
```

yuvenus, 2024-07-22:
Hi @Prev-I! Glad to see a fellow Windows 11 user 😄 I've found the issue you described happens to me often when switching branches/versions, and doing a "fresh" install does the trick, like so:

```
npm run fresh 
npm run install-all
```

`npm run fresh` will remove all built files and node modules folders, and `npm run install-all` will iterate and install everything again.

I did this on my machine before writing comment and confirmed that `docker compose build` built successfully 👍 Please let me know how running the above commands go and if you have any more questions, thank you!

Prev-I, 2024-07-23:
Hi @yuvenus, you are right, deleting folers with npm run fresh resolved the issue.

It could be usefull to add it to the docker compose to be executed before build?

yuvenus, 2024-07-23:
> Hi @yuvenus, you are right, deleting folers with npm run fresh resolved the issue.
> 
> It could be usefull to add it to the docker compose to be executed before build?

That's a good point! 

If you feel comfortable, please open a PR that adds the fresh/install-all commands into the docker compose -- otherwise we'll make an internal note to prioritize and do this 👍 Please let us know if it makes sense to close this ticket as well, given this request. Thank you!

Prev-I, 2024-07-26:
@yuvenus  I was preparing the pull request when I've noted that the build are using the command:
`npm i `
instead of the reproducible counterpart
'npm ci'

the second one take care of deleting the node_module folder and restore exactly the module version reported in the package-lock.json preventing strange situation where some sub-dependecies have different version on different machines.

I'll make a pull request that will add the npm ci command and the package lock files for both ocpi and core project. 

From what branch I should start? main? and where a should request the merge to?

yuvenus, 2024-07-26:
@Prev-I Amazing, thank you!!

Our current strategy is to branch off `main`, but open a PR to the next RC, which in this case would be `rc-1.3.2`. Please continue with this strategy, and we can retarget as needed 👍

Prev-I, 2024-07-28:
@yuvenus I've tried to push a feature branch but my user is not authorized :(

closed 2025-01-06 by masonseaman