citrineos/citrineos #34: Can't get past Citrine Build 4/5

arahman33, 2024-08-01:
I want to start by saying I never had any problem building CitrineOS (for docker). However, since yesterday my build keeps getting stuck at npm install --workspaces --verbose && npm run compile --workspaces --verbose. Every npm fetch returns cache miss. I suspected it may be a network issue so I changed to 3 different networks. I also tried to build in a different device (incase it is the device!). Anyone facing the same issue please help.

More info: npm -v 10.8.2 node -v 20.16.0

62.72 npm http fetch GET 200 https://registry.npmjs.org/source-map-js/-/source-map-js-1.2.0.tgz 3229ms (cache miss)
62.77 npm http fetch GET 200 https://registry.npmjs.org/@vue/reactivity/-/reactivity-3.4.35.tgz 3275ms (cache miss)
62.77 npm http fetch GET 200 https://registry.npmjs.org/has-bigints/-/has-bigints-1.0.2.tgz 3257ms (cache miss)
62.77 npm http fetch GET 200 https://registry.npmjs.org/which-boxed-primitive/-/which-boxed-primitive-1.0.2.tgz 3260ms (cache miss)
.
.
.
78.00 npm http fetch GET 200 https://registry.npmjs.org/typescript/-/typescript-5.0.4.tgz 17669ms (cache miss)
78.47 npm http fetch GET 200 https://registry.npmjs.org/typescript/-/typescript-5.0.4.tgz 18139ms (cache miss)
78.61 npm http fetch GET 200 https://registry.npmjs.org/typescript/-/typescript-5.5.4.tgz 18290ms (cache miss)
78.95 npm http fetch GET 200 https://registry.npmjs.org/prettier/-/prettier-2.8.8.tgz 18469ms (cache miss)
79.04 npm info run bcrypt@5.1.1 install node_modules/bcrypt node-pre-gyp install --fallback-to-build
79.07 npm info run bufferutil@4.0.8 install node_modules/bufferutil node-gyp-build
79.08 npm info run deasync@0.1.30 install node_modules/deasync node ./build.js
79.09 npm info run sqlite3@5.1.7 install node_modules/sqlite3 prebuild-install -r napi || node-gyp rebuild
79.11 npm info run utf-8-validate@6.0.4 install node_modules/utf-8-validate node-gyp-build
79.53 npm info run bufferutil@4.0.8 install { code: 0, signal: null }
79.53 npm info run utf-8-validate@6.0.4 install { code: 0, signal: null }
79.62 npm info run deasync@0.1.30 install { code: 0, signal: null }
80.36 npm info run bcrypt@5.1.1 install { code: 0, signal: null }
80.98 npm info run sqlite3@5.1.7 install { code: 0, signal: null }
80.98 npm info run es5-ext@0.10.64 postinstall node_modules/es5-ext node -e "try{require('./_postinstall')}catch(e){}" || exit 0
80.99 npm info run protobufjs@7.2.4 postinstall node_modules/protobufjs node scripts/postinstall
81.00 npm info run protobufjs@7.3.2 postinstall node_modules/@grpc/proto-loader/node_modules/protobufjs node scripts/postinstall
81.01 npm info run vue-demi@0.14.10 postinstall node_modules/pinia/node_modules/vue-demi node -e "try{require('./scripts/postinstall.js')}catch(e){}"
81.06 npm info run protobufjs@7.2.4 postinstall { code: 0, signal: null }
81.07 npm info run protobufjs@7.3.2 postinstall { code: 0, signal: null }
81.11 npm info run es5-ext@0.10.64 postinstall { code: 0, signal: null }
81.17 npm info run vue-demi@0.14.10 postinstall { code: 0, signal: null }
81.91
81.91 added 851 packages, and audited 864 packages in 1m
81.91
81.91 160 packages are looking for funding
81.91 run npm fund for details
81.93
81.93 6 vulnerabilities (3 moderate, 3 critical)
81.93
81.93 To address all issues (including breaking changes), run:
81.93 npm audit fix --force
81.93
81.93 Run npm audit for details.
81.93 npm notice
81.93 npm verbose exit 0
81.93 npm info ok
82.27 npm verbose cli /usr/local/bin/node /usr/local/bin/npm
82.27 npm info using node@v18.20.4
82.28 npm verbose title npm run compile
82.28 npm verbose argv "run" "compile" "--workspaces" "--loglevel" "verbose"
82.28 npm verbose logfile logs-max:10 dir:/root/.npm/_logs/2024-08-01T15_29_48_289Z-
82.28 npm verbose logfile /root/.npm/_logs/2024-08-01T15_29_48_289Z-debug-0.log
82.40
82.40 > @citrineos/base@1.1.1 compile
82.40 > tsc -p tsconfig.json
82.40
88.32
88.32 > @citrineos/data@1.1.1 compile
88.32 > tsc -p tsconfig.json
88.32
91.96
91.96 > @citrineos/util@1.1.1 compile
91.96 > tsc -p tsconfig.json
91.96
97.60 src/util/swagger.ts(83,10): error TS2769: No overload matches this call.
97.60 Overload 1 of 3, '(plugin: FastifyPluginCallback<SwaggerOptions, RawServerDefault, FastifyTypeProviderDefault, FastifyBaseLogger>, opts?: FastifyRegisterOptions<...> | undefined): FastifyInstance<...> & PromiseLike<...>', gave the following error.
97.60 Type '({ swaggerObject, openapiObject, }: { swaggerObject: Partial<Document<{}>>; openapiObject: Partial<OpenAPIV3.Document<{}> | OpenAPIV3_1.Document<{}>>; }) => Partial<...>' is not assignable to type 'SwaggerTransformObject'.
97.60 Types of parameters '__0' and 'documentObject' are incompatible.
97.60 Type 'SwaggerDocumentObject' is not assignable to type '{ swaggerObject: Partial<Document<{}>>; openapiObject: Partial<OpenAPIV3.Document<{}> | OpenAPIV3_1.Document<{}>>; }'.
97.60 Property 'openapiObject' is missing in type '{ swaggerObject: Partial<Document<{}>>; }' but required in type '{ swaggerObject: Partial<Document<{}>>; openapiObject: Partial<OpenAPIV3.Document<{}> | OpenAPIV3_1.Document<{}>>; }'.
97.60 Overload 2 of 3, '(plugin: FastifyPluginAsync<SwaggerOptions, RawServerDefault, FastifyTypeProviderDefault, FastifyBaseLogger>, opts?: FastifyRegisterOptions<...> | undefined): FastifyInstance<...> & PromiseLike<...>', gave the following error.
97.60 Argument of type 'FastifySwagger' is not assignable to parameter of type 'FastifyPluginAsync<SwaggerOptions, RawServerDefault, FastifyTypeProviderDefault, FastifyBaseLogger>'.
97.60 Target signature provides too few arguments. Expected 3 or more, but got 2.
97.60 Overload 3 of 3, '(plugin: FastifyPluginCallback<SwaggerOptions, RawServerDefault, FastifyTypeProviderDefault, FastifyBaseLogger> | FastifyPluginAsync<...> | Promise<...> | Promise<...>, opts?: FastifyRegisterOptions<...> | undefined): FastifyInstance<...> & PromiseLike<...>', gave the following error.
97.60 Type '({ swaggerObject, openapiObject, }: { swaggerObject: Partial<Document<{}>>; openapiObject: Partial<OpenAPIV3.Document<{}> | OpenAPIV3_1.Document<{}>>; }) => Partial<...>' is not assignable to type 'SwaggerTransformObject'.
97.66 npm error Lifecycle script compile failed with error:
97.66 npm error Error: command failed
97.66 npm error in workspace: @citrineos/util@1.1.1
97.66 npm error at location: /usr/local/apps/citrineos/02_Util
97.71
97.71 > @citrineos/certificates@1.1.1 compile
97.71 > tsc -p tsconfig.json
97.71
100.6
100.6 > @citrineos/configuration@1.1.1 compile
100.6 > tsc -p tsconfig.json
100.6
104.1
104.1 > @citrineos/evdriver@1.1.1 compile
104.1 > tsc -p tsconfig.json
104.1
107.1
107.1 > @citrineos/monitoring@1.1.1 compile
107.1 > tsc -p tsconfig.json
107.1
110.1
110.1 > @citrineos/ocpprouter@1.1.1 compile
110.1 > tsc -p tsconfig.json
110.1
112.3
112.3 > @citrineos/reporting@1.1.1 compile
112.3 > tsc -p tsconfig.json
112.3
115.2
115.2 > @citrineos/smartcharging@1.1.1 compile
115.2 > tsc -p tsconfig.json
115.2
117.9
117.9 > @citrineos/transactions@1.1.1 compile
117.9 > tsc -p tsconfig.json
117.9
121.2
121.2 > @citrineos/server@1.1.1 compile
121.2 > tsc -p tsconfig.json
121.2
124.5 npm verbose exit 1
124.5 npm verbose code 1

failed to solve: process "/bin/sh -c npm install --workspaces --verbose && npm run compile --workspaces --verbose" did not complete successfully: exit code: 1

ChrisWeissmann, 2024-08-02:
Hi @arahman33, it looks like the issue you are running into is due to an update of the dependency Fastify/swagger. We fixed it in this commit by pinning the version https://github.com/citrineos/citrineos-core/commit/ca2fae5a283a147f589ef7ad616796c67a5a9c69

The latest from main has the commit in it.

thanaParis, 2024-09-18:
Closing this as fixed.

closed 2024-09-18 by thanaParis