citrineos/citrineos #37: Put method failed for configuration/boot api

sithu-go, 2024-08-19:
**Repository**

- citrineos-core

**Describe the bug**
The API currently does not support the lastBootTime query parameter, and there is no option to pass lastBootTime in the request body. This limitation prevents users from fully interacting with the API as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Call this api
```sh
curl --location --request PUT 'http://localhost:8080/data/configuration/boot?stationId=cp001' \
--header 'Content-Type: application/json' \
--data '{ "status": "Pending" }'
```
2. try to add lastBootTime, It will respond the same error.
```sh
curl --location --request PUT 'http://localhost:8080/data/configuration/boot?stationId=cp001' \
--header 'Content-Type: application/json' \
--data '{ "status": "Pending", "lastBootTime": "2024-08-19" }'
```

Error logs 
```log
ERROR   /usr/local/apps/citrineos-core/00_Base/dist/interfaces/api/AbstractModuleApi.js:119     CitrineOS Logger:ConfigurationModuleApi Error in handling data route

TypeError  Cannot read properties of undefined (reading 'toISOString')
error stack:
  • Boot.js     Boot.get
        /usr/local/apps/citrineos-core/01_Data/dist/layers/sequelize/model/Boot.js:33
  • model.js    Boot.get
        /usr/local/apps/citrineos-core/node_modules/sequelize/lib/model.js:2168
  • model.js    Boot.get
        /usr/local/apps/citrineos-core/node_modules/sequelize/lib/model.js:2190
  • model.js    Boot.update
        /usr/local/apps/citrineos-core/node_modules/sequelize/lib/model.js:1929
  • model.js    Model.<computed> [as update]
        /usr/local/apps/citrineos-core/node_modules/sequelize-typescript/dist/model/model/model.js:128
  • Base.js     SequelizeBootRepository.<anonymous>
        /usr/local/apps/citrineos-core/01_Data/dist/layers/sequelize/repository/Base.js:98
  • 
  • Base.js
        /usr/local/apps/citrineos-core/01_Data/dist/layers/sequelize/repository/Base.js:12
  • 
  • Base.js     __awaiter
        /usr/local/apps/citrineos-core/01_Data/dist/layers/sequelize/repository/Base.js:8

```

**Expected behavior**
The API should support the `lastBootTime` parameter in the query string or request body, allowing for the full expected interaction with the data.

yuvenus, 2024-08-19:
Thank you for creating this issue @sithu-go! We've identified the issue and created a PR here: https://github.com/citrineos/citrineos-core/pull/238

Though we won't add a `lastBootTime` in the query params nor the request body, since that field automatically updates on its own as the charger boots, this fix should allow `PUT /data/configuration/boot` to complete properly.

thanaParis, 2024-09-18:
This fix has been released!