citrineos/citrineos #38: Swagger UI shows 0.0.0.0 in "Try it out" instead of server URL

marck-brusa, 2024-08-27:
**Which repository?**
Is it:
- citrineos-core

**Describe the bug**

The SwaggerUI in rc-1.4.2 sends the "try it out" URL as 0.0.0.0, when running from the docker container. 

Our setup has the Citrine Swaagger API service running on a dedicate VM, with a FQDN, when accessing the url:

https://citrine.company.com:8080/docs

or via a reverse proxy 

https://citrine-api.company.com/docs

any of the "try it out" links incorrectly show a http://0.0.0.0/, despite the base URL being correctly detected at the top of the Swagger UI.

I tried editing variable for host in centralSystem for the file  .../Server/src/config/envs/docker.ts file to indicate the hostname, or the IP address of my VM, however the application is not able to start, since it binds to the internal Docker network, not my actual IP address.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Citrine from docker on a machine that is remote (not accessing via localhost)
2. Visit any of the Swagger UI elements, and use the "Try it out"
3. URL is returned at 0.0.0.0

**Expected behavior**
The URL should be returned as 

http://some.url.here/docs

instead of http:/0.0.0.0/docs

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome 128.0.6613.84
 - Version 128.0.6613.84
(Same behavior in Firefox 129.0.2)

**Additional context**
Maybe have a look here:

https://swagger.io/docs/specification/api-host-and-base-path/

yuvenus, 2024-09-04:
Hello @marck-brusa, nice catch! Thank you for reporting this issue.

If you feel comfortable, please open a PR to resolve this following [our contribution guidelines](https://github.com/citrineos/citrineos/blob/main/CONTRIBUTING.md). Otherwise we'll take it internally and keep you posted on when this is resolved.

thanaParis, 2024-09-18:
This has been fixed as of 1.4.3!

closed 2024-09-18 by thanaParis