citrineos/citrineos #45: citrineos-core running failure in local system(macOS)

unaccepted, 2024-10-23:
**Which repository?**

- citrineos-core

**Describe the bug**
I am trying to run citrineos-core server, but facing multiple issues described below.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the citrine-core repo.
2. Install Node (v18.20.4)
3. Install NPM (10.7.0)
4. Install Docker(Docker version 24.0.5, build ced0996)
5. npm run install-all
6. npm run build
7. cd server
8. docker-compose -f ./docker-compose.yml up -d
9. npm run start

**Expected behavior**
CitrineOS-core server should be up and running according to the readme.

**Screenshots**
below screenshot is after 'npm run build'. Not sure whether its actually successful build or not:
[image]

below screenshot is after 'docker-compose -f ./docker-compose.yml up -d'
[image]

below screenshot is after 'npm run start'
[image]

MacOS version: 12.5

unaccepted, 2024-10-25:
@thanaParis can anyone help on this setup issue please.

thanaParis, 2024-10-25:
Hi @unaccepted !

It looks like there may be some issues with your local node environment. Here are some threads I found after searching for your errors on google:

https://github.com/nodejs/help/issues/3686

https://stackoverflow.com/questions/27541586/error-could-not-locate-the-bindings-file-tried-56

Following some of the advice above should help you get things set up correctly.
Both of these issues appear to be related to node-gyp. It may be you have an outdated version of it locally, or it was installed incorrectly. If problems persist despite the advice in the links above, I would consider making sure you start from scratch and reinstall the node dependencies.

unaccepted, 2024-11-05:
Hi @thanaParis Thanks for the response.
I tried multiple times, even on another system. But I am getting same set of errors posted above.
Note that: I am using macOS and python3 version is: 3.9.6

Can you tell me the exact versions of node, npm, python3 to be used.
Also, do we need to change anything in the package.js or some other files to run it locally.

unaccepted, 2024-11-05:
@thanaParis 
Please find all the versions.
[image]

[image]

How to resolve this. This I am getting when hitting "docker-compose -f ./docker-compose.yml up -d"

cclauss, 2025-03-19:
> 125.6 npm error gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"

Python is not the problem because the line above says that a supported version of Python was found at `/usr/bin/python3`.