citrineos/citrineos #48: Everest Not running on Mac silicon **no matching manifest for linux/arm64/v8 in the manifest list entries**

hpxix, 2024-10-28:
**Which repository?**
Is it:

- citrineos-core/server/everest

**Describe the bug**
Description: Running the EVerest charger simulator on Mac Silicon (M1/M2) presents compatibility issues due to differences in architecture, as Docker defaults to pulling AMD64 images that are incompatible with ARM64 processors. This causes errors when attempting to start the containers in the /Server/everest directory using the docker-compose.yml setup.

**ERROR**
no matching manifest for linux/arm64/v8 in the manifest list entries

**Proposed Solution:**
To ensure compatibility and allow EVerest to run smoothly on Mac Silicon, we propose the following adjustments to the docker-compose.yml file:

**Platform Specification:**
Add platform: linux/amd64 under each service definition to enforce Docker’s usage of AMD64 images with ARM64 emulation on Apple Silicon.**

**Direct Environment Variables:**
For simpler configuration, define EVEREST_IMAGE_TAG and EVEREST_TARGET_URL directly in the environment section within docker-compose.yml. This avoids needing dynamic build-time arguments and simplifies setup.

**Updated docker-compose.yml Example aka FIX:**
[image]
[image]

**Steps to Reproduce:**
Clone the EVerest repository and navigate to /Server/everest.
Run npm run start-everest or npm run start-everest-windows to start the Docker containers.
Observe any errors related to architecture incompatibility.

**Expected Outcome:**
With the platform: linux/amd64 specification, Docker should emulate AMD64, allowing the containers to run seamlessly on Mac Silicon. Users should then be able to access the EVerest UI at [localhost|ip]:1880/ui/ and view OCPP logs at localhost:8888.

**Additional Context:**
 This fix aims to support ARM64 users by simplifying the setup and ensuring all components are appropriately configured for cross-platform compatibility.

**I'll commit a new Readme.md for EVerest for Mac Users later today have a great one!**

hpxix, 2024-10-28:
I've included a new readme.md for EVerest in /server/EVerest/readme.md regarding running Everest with Mac arm64 settings:
https://github.com/citrineos/citrineos-core/pull/286

@thanaParis Please approve for merging, thank you💙.

corrosive4354, 2024-10-29:
Check if the image supports ARM64:
If possible, check if there is an ARM64 version of the EVerest image. If so, using that version may be more efficient.

If it still doesn't work,
There are too many compatibility issues between docker and mac. You can use [Servbay](https://servbay.com) instead of docker to deploy the environment. Servbay is more professional on mac.

elliot-sabitov, 2024-10-29:
@hpxix we were looking deeper into this as well, and we did see that initially on ARM64 we see the `no matching manifest for linux/arm64/v8 in the manifest list entries` error and when we add `platform=linux/amd64` then it does move past that error, but when `manager` starts up, it just immediately fails with:

```
2024-05-24 18:26:39.489204 [ERRO] manager         int main(int, char**) :: Main manager process exits because of caught exception:
Syscall pipe2() failed (Invalid argument), exiting
```

Were you able to get around this and actually have Everest running?

elliot-sabitov, 2024-10-29:
I followed the changes that you have made in the video https://www.loom.com/share/e7fd054e4d224918a722d14eb834665b and I am seeing the above error (`Syscall pipe2() failed`). 

I am on a 2021 Macbook Pro with Apple M1 Max Chip. I tried running via `npm run start-everest` and manually running the compose command. It seems that no matter what I do, I end up seeing this error and the manager exists immediately... 

[image]

hpxix, 2024-10-29:
Actually my PR didn't include all changes but if you go through my version of the fix for the issue above it'll work,

**Main changes that are done;**

add platform tag = linux/amd64 to all services,
add environment variables in manager service that are 
`-EVEREST_TARGET_URL=ws://EVEREST_TARGET_URL=ws://host.docker.internal:8081/cp001`
`- EVEREST_IMAGE_TAG=0.0.16`

hpxix, 2024-10-29:
@thanaParis I've added a new PR with the latest changes on the readme.md for server/everest:
https://github.com/citrineos/citrineos-core/pull/286/commits/eceaaec97464cfb43e85b2bab7108dd3ea2771e2

elliot-sabitov, 2024-10-29:
@hpxix I tried with the following 2 files just now (see below), but unfortunately I am still seeing this error `Syscall pipe2() failed (Invalid argument), exiting` - I am wondering what else may be different on my machine vs yours and I am very eager to get this working on our M1 machines and hope that we can get this resolved! Thank you so much for your help!
```
// docker-compose.xml
version: '3.8'
services:
  mqtt-server:
    image: ghcr.io/everest/everest-demo/mqtt-server:0.0.16
    platform: linux/amd64
    networks:
      - everest-net
    logging:
      driver: none
  manager:
    build:
      context: .
      dockerfile: Dockerfile
    platform: linux/amd64
    ports:
      - 8888:8888
    networks:
      - everest-net
    depends_on:
      - mqtt-server
    environment:
      - MQTT_SERVER_ADDRESS=mqtt-server
      - EVEREST_TARGET_URL=ws://host.docker.internal:8081/cp001
      - EVEREST_IMAGE_TAG=0.0.16
    sysctls:
      - net.ipv6.conf.all.disable_ipv6=0
    extra_hosts:
      - 'host.docker.internal:host-gateway'
  nodered:
    image: ghcr.io/everest/everest-demo/nodered:0.0.16
    platform: linux/amd64
    networks:
      - everest-net
    depends_on:
      - mqtt-server
    ports:
      - 1880:1880
    environment:
      - MQTT_SERVER_ADDRESS=mqtt-server
      - FLOWS=/config/config-sil-two-evse-flow.json
networks:
  everest-net:
    driver: bridge
```

and

```
// Dockerfile
ARG EVEREST_IMAGE_TAG=0.0.16
FROM --platform=linux/amd64 ghcr.io/everest/everest-demo/manager:${EVEREST_IMAGE_TAG}
ARG EVEREST_TARGET_URL=ws://host.docker.internal:8081/cp001
ENV EVEREST_TARGET_URL $EVEREST_TARGET_URL
WORKDIR /workspace
RUN ["/entrypoint.sh"]
RUN apk update && apk add sqlite
RUN sqlite3 /ext/source/build/dist/share/everest/modules/OCPP201/device_model_storage.db \
        "UPDATE VARIABLE_ATTRIBUTE \
        SET value = '[{\"configurationSlot\": 1, \"connectionData\": {\"messageTimeout\": 30, \"ocppCsmsUrl\": \"$EVEREST_TARGET_URL\", \"ocppInterface\": \"Wired0\", \"ocppTransport\": \"JSON\", \"ocppVersion\": \"OCPP20\", \"securityProfile\": 1}},{\"configurationSlot\": 2, \"connectionData\": {\"messageTimeout\": 30, \"ocppCsmsUrl\": \"$EVEREST_TARGET_URL\", \"ocppInterface\": \"Wired0\", \"ocppTransport\": \"JSON\", \"ocppVersion\": \"OCPP20\", \"securityProfile\": 1}}]' \
        WHERE \
        variable_Id IN ( \
        SELECT id FROM VARIABLE \
        WHERE name = 'NetworkConnectionProfiles' \
        );"
RUN rm /ext/source/build/dist/etc/everest/certs/ca/v2g/*.*
RUN npm i -g http-server
EXPOSE 8888
COPY ./start.sh /tmp/start.sh
RUN chmod +x /tmp/start.sh
CMD ["sh", "-c", "/tmp/start.sh"]
```

elliot-sabitov, 2024-10-29:
Perhaps we have different versions of docker installed?? [image]

corrosive4354, 2024-10-30:
To ensure that EVerest runs smoothly on Mac Silicon, you can make the following adjustments to the docker-compose.yml file:

Add platform: linux/amd64 under each service definition to force Docker to use the AMD64 image and emulate on ARM64.

Define EVEREST_IMAGE_TAG and EVEREST_TARGET_URL directly in the environment section of docker-compose.yml to avoid the parameter requirements during dynamic builds and simplify configuration.

Or You can use [servbay](servbay.com) instead of docker, which is more professional on mac and has fewer bugs.

hpxix, 2024-10-30:
@elliot-sabitov 
This is the docker Version I'm using considering updating your docker desktop as I faced issues when running citrineos because of my older version but mine was 3.x.. although it's fair to mention that docker version plays a big part of this, updating your docker and you should be able to run Everest manager and if not please show me the error you're encountering so we can look through it even deeper.

Here are things I'd consider if i was facing that problem,
First check the image compatibility of the architecture by using
- `docker image inspect ghcr.io/everest/everest-demo/mqtt-server:0.0.16`
this command works even if the image is not currently running the expected response should be 
        "Architecture": "amd64", 

If it wasn't running try pulling it using:
docker pull ghcr.io/everest/everest-demo/mqtt-server:0.0.16

also try running the image using:
`docker run --platform=linux/arm64 your_image_name`

[image]

elliot-sabitov, 2024-10-30:
Hi @hpxix , when I run `docker image inspect ghcr.io/everest/everest-demo/mqtt-server:0.0.16` I see:
```
[
    {
        "Id": "sha256:7f1f90da85807589920007742b650bbd244f7b4c3c5118784d47526f852f944b",
        "RepoTags": [
            "ghcr.io/everest/everest-demo/mqtt-server:0.0.16"
        ],
        "RepoDigests": [
            "ghcr.io/everest/everest-demo/mqtt-server@sha256:fca65b218859f28290e2a23e8ee37ffbff80c079bd47a533e388bf58b78757f8"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2024-09-18T00:25:00.201720761Z",
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "1883/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "VERSION=2.0.10",
                "DOWNLOAD_SHA256=0188f7b21b91d6d80e992b8d6116ba851468b3bd154030e8a003ed28fb6f4a44",
                "GPG_KEYS=A0D6EEA1DCAE49A635A3B2F0779B22DFB3E717B7",
                "LWS_VERSION=2.4.2",
                "LWS_SHA256=73012d7fcf428dedccc816e83a63a01462e27819d5537b8e0d0c7264bfacfad6",
                "CJSON_VERSION=1.7.14",
                "CJSON_SHA256=fb50a663eefdc76bafa80c82bc045af13b1363e8f45cec8b442007aef6a41343"
            ],
            "Cmd": [
                "/usr/sbin/mosquitto",
                "-c",
                "/mosquitto/config/mosquitto.conf"
            ],
            "Image": "",
            "Volumes": {
                "/mosquitto/data": {},
                "/mosquitto/log": {}
            },
            "WorkingDir": "",
            "Entrypoint": [
                "/docker-entrypoint.sh"
            ],
            "OnBuild": null,
            "Labels": {
                "description": "Eclipse Mosquitto MQTT Broker",
                "maintainer": "Roger Light <roger@atchoo.org>",
                "org.opencontainers.image.created": "2024-09-18T00:24:57.475Z",
                "org.opencontainers.image.description": "EVerest demo: Dockerized demo with software in the loop simulation",
                "org.opencontainers.image.licenses": "Apache-2.0",
                "org.opencontainers.image.revision": "79d0f94ec8d6c3cd7088740f445828f956d2c8e3",
                "org.opencontainers.image.source": "https://github.com/EVerest/everest-demo",
                "org.opencontainers.image.title": "everest-demo",
                "org.opencontainers.image.url": "https://github.com/EVerest/everest-demo",
                "org.opencontainers.image.version": "0.0.16"
            }
        },
        "Architecture": "amd64",
        "Os": "linux",
        "Size": 9828369,
        "GraphDriver": {
            "Data": {
                "LowerDir": "/var/lib/docker/overlay2/351f09a1eaab2c37554e50a467f826ce6ef27d8626826d9bde7dd7f9795d1256/diff:/var/lib/docker/overlay2/7389e22cf99077758878f5ba70607a1f12b16c854bc19e381e4edd658fc594ea/diff:/var/lib/docker/overlay2/d42f103fcbb59bc597cb5624064977b204da6e18f7f304b6400a9e3c52db80c2/diff",
                "MergedDir": "/var/lib/docker/overlay2/f46405c31be0d627362cec2bf5911f8603e2f68cda6f36d65b138ec0c7eadc7f/merged",
                "UpperDir": "/var/lib/docker/overlay2/f46405c31be0d627362cec2bf5911f8603e2f68cda6f36d65b138ec0c7eadc7f/diff",
                "WorkDir": "/var/lib/docker/overlay2/f46405c31be0d627362cec2bf5911f8603e2f68cda6f36d65b138ec0c7eadc7f/work"
            },
            "Name": "overlay2"
        },
        "RootFS": {
            "Type": "layers",
            "Layers": [
                "sha256:32f366d666a541852cad754ee1cdb53a736110b550f0c2d5a46bc5ba519896b6",
                "sha256:ec8e0c7af5819c3731ae879aabb1d33f45661c9b5ef9e7c7f8516d9a4e4f5a49",
                "sha256:0cb04d470e214b32f6f44c894e6f99626304b9f1cdcc9b1c04491f5967d75d4f",
                "sha256:8d610432c3ac23d5c0c2b4495c66d4a751aea743697208b1d3a412873708770a"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```
When I run `uname -m` I get `arm64`.

@hpxix Can you please try running `uname -m` to confirm that you are also seeing `arm64`? 🙏 

I did upgrade my docker to the latest version 4.34.3 but I am still seeing the same behavior, as soon as the Everest manager container starts up, it exists, and checking the logs, I still see the following output:
```
2024-10-30 12:43:04 2024-10-30 16:43:04.443816 [INFO] manager          ::   ________      __                _   
2024-10-30 12:43:04 2024-10-30 16:43:04.449232 [INFO] manager          ::  |  ____\ \    / /               | |  
2024-10-30 12:43:04 2024-10-30 16:43:04.449283 [INFO] manager          ::  | |__   \ \  / /__ _ __ ___  ___| |_ 
2024-10-30 12:43:04 2024-10-30 16:43:04.449583 [INFO] manager          ::  |  __|   \ \/ / _ \ '__/ _ \/ __| __|
2024-10-30 12:43:04 2024-10-30 16:43:04.449597 [INFO] manager          ::  | |____   \  /  __/ | |  __/\__ \ |_ 
2024-10-30 12:43:04 2024-10-30 16:43:04.449609 [INFO] manager          ::  |______|   \/ \___|_|  \___||___/\__|
2024-10-30 12:43:04 2024-10-30 16:43:04.449620 [INFO] manager          :: 
2024-10-30 12:43:04 2024-10-30 16:43:04.449659 [INFO] manager          :: Using MQTT broker mqtt-server:1883
2024-10-30 12:43:04 2024-10-30 16:43:04.467492 [ERRO] manager         int main(int, char**) :: Main manager process exits because of caught exception:
2024-10-30 12:43:04 Syscall pipe2() failed (Invalid argument), exiting
```

elliot-sabitov, 2024-11-01:
To provide an update, I had another developer using Macbook Pro with M1 chip try the above, and she also experienced the same issue `Syscall pipe2() failed`. We are still looking into how you are able to run this on your Macbook Air running the same `arm64` architecture.

Thank you in advance for your patience.

fabiolnm, 2025-01-10:
Update:

- It works with a M1  🙌
- The original issue is still happening with a M2:

```
citrineos-core % cd Server/everest 

Found '/Users/fabiomiranda/csms/citrineos-core/Server/.nvmrc' with version <v18.10.0>
Now using node v18.10.0 (npm v8.19.2)
fabiomiranda@Fabios-MacBook-Pro everest % npm run start-everest

> @citrineos/server@1.5.0 start-everest
> cd ./everest && cross-env EVEREST_IMAGE_TAG=0.0.16 EVEREST_TARGET_URL=ws://host.docker.internal:8081/cp001 docker-compose up

WARN[0000] /Users/fabiomiranda/csms/citrineos-core/Server/everest/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion 
[+] Running 0/6
 ⠙ nodered Pulling                                                                                                                    1.1s 
 ⠙ mqtt-server [⠀⠀⠀⠀] Pulling                                                                                                         1.1s 
   ⠹ 339de151aab4 Pulling fs layer                                                                                                    0.2s 
   ⠹ 8cc7b833e2db Pulling fs layer                                                                                                    0.2s 
   ⠹ f2b454175e7e Pulling fs layer                                                                                                    0.2s 
   ⠹ e41a60e4211c Waiting                                                                                                             0.2s 
no matching manifest for linux/arm64/v8 in the manifest list entries
npm ERR! Lifecycle script `start-everest` failed with error: 
npm ERR! Error: command failed 
npm ERR!   in workspace: @citrineos/server@1.5.0 
npm ERR!   at location: /Users/fabiomiranda/csms/citrineos-core/Server 
```

elliot-sabitov, 2025-01-29:
Great news, we were able to get Everest to run on an M1 machine! We used the latest `rc-1.6.0` branch, but it should work using prior branches, or the example code samples above. 

The deciding factor was the `Use Rosetta for x86_64/amd64 emulation on Apple Silicon` check box in Docker Desktop > Settings > Virtual Machine Options (see screenshot). Disabling this option, eliminates the `Syscall pipe2() failed` error and we were able to use latest `rc-1.6.0` without any additional changes to run Everest!

[image]
[image]

closed 2025-01-30 by thanaParis