citrineos/citrineos #61: Invalid BootNotification Causes CitrineOS to Crash

brandonprry, 2024-11-23:
**Which repository?**
Is it:

- citrineos-core

**Describe the bug**

Sending an incomplete BootNotification causes CitrineOS to crash.

**To Reproduce**

Send the following boot notification

```
[2, "5e58c16f-32ee-4589-ae5d-2272e5beeb71", "BootNotification", {}]
```

CitrineOS crashes after responding with the following stack trace.

```
2024-11-22 18:57:26 /usr/local/apps/citrineos/03_Modules/Configuration/dist/module/DeviceModelService.js:88
2024-11-22 18:57:26                     value: chargingStation.model,
2024-11-22 18:57:26                                            ^
2024-11-22 18:57:26 
2024-11-22 18:57:26 TypeError: Cannot read properties of undefined (reading 'model')
2024-11-22 18:57:26     at DeviceModelService.<anonymous> (/usr/local/apps/citrineos/03_Modules/Configuration/dist/module/DeviceModelService.js:88:44)
2024-11-22 18:57:26     at Generator.next (<anonymous>)
2024-11-22 18:57:26     at /usr/local/apps/citrineos/03_Modules/Configuration/dist/module/DeviceModelService.js:11:71
2024-11-22 18:57:26     at new Promise (<anonymous>)
2024-11-22 18:57:26     at __awaiter (/usr/local/apps/citrineos/03_Modules/Configuration/dist/module/DeviceModelService.js:7:12)
2024-11-22 18:57:26     at DeviceModelService.updateDeviceModel (/usr/local/apps/citrineos/03_Modules/Configuration/dist/module/DeviceModelService.js:82:16)
2024-11-22 18:57:26     at ConfigurationModule.<anonymous> (/usr/local/apps/citrineos/03_Modules/Configuration/dist/module/module.js:136:38)
2024-11-22 18:57:26     at Generator.next (<anonymous>)
2024-11-22 18:57:26     at fulfilled (/usr/local/apps/citrineos/03_Modules/Configuration/dist/module/module.js:18:58)
2024-11-22 18:57:26     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
2024-11-22 18:57:26 
```

**Expected behavior**

It doesn't crash

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: OSX/Docker
 - Browser Firefox

**Additional context**
Add any other context about the problem here.

brandonprry, 2024-12-03:
Is there a better place to report such issues?

masonseaman, 2025-01-06:
Nope this is the best place to report issues! There is a PR to fix this in an upcoming RC

closed 2025-06-09 by thanaParis