citrineos/citrineos #123: kWh not summing correctly for some charging sessions

masonseaman, 2025-03-10:
**Which repository?**
Is it:
- citrineos-core

**Describe the bug**
We only count energy.active.import.register measurands that do not have a phase, we shouldn't limit to only these measurands.

**Additional context**
Add any other context about the problem here.

stfnrm, 2025-04-13:
Hey @masonseaman 
I came across the thread in Discord about this issue.
I think I have the same issue with our charging station as it is always shown as "0" although the car is charging. 

What is the status of this ticket?
Is there any way to push the fix forward?

Thanks!
Stefan

masonseaman, 2025-04-17:
hey @stfnrm!

Sorry for the late reply, we have a PR up to resolve this and the fix should be in main soon!