citrineos/citrineos #142: OCPI setup issue

testocpp, 2025-06-24:
Hi 

Thanks for setting up a really good tool to enable OCPP learning. I was able to setp dockers for the core tool and was able to access UI docker also. 

Somehow ran into following error while setting up docker for OCPI when I ran docker compose command in OCPI/server folder. Any help would be appreciated. 

citrine-1      | 2025-06-24 13:47:07.558        ERROR   /usr/local/apps/citrineos/02_Util/dist/files/s3Storage.js:93    S3Storage       Error fetching config from S3:
citrine-1      | 
citrine-1      |  Error  Region is missing 

Cheers

testocpp, 2025-06-24:
Futher to above based on some of the issue which were previously resolved I did following

1. Core : Ran
 npm run fresh 
npm run install-all
2. OCPI directory : Ran following commands 
npm run fresh 
npm run install-all
3. Core/Server: Ran 
docker compose 
4. All docker containers are up and running but when I run UI docker compose command 
5. All necessary containers seem tobe running but not sure how to access OCPI on localhost:3000 ?

testocpp, 2025-06-26:
Hi @masonseaman , thanks for taking out time and   providing clarification. I have been scratching my head for last couple of days on what I may be missing. 

Is there a version of core and ocpi where the two work together?

masonseaman, 2025-06-26:
core 1.5 should work with the latest ocpi