citrineos/citrineos #149: Use of Kafka as message broker

Aditya-agra, 2025-07-14:
I wanted to use the kafka broker in the project instead of RabbitMq. I added kafka in the docker-compose.yml file and ran it in my local system.

[image]

All the container seems to come up, but i wasn't able to make the websocket connection. Is there some documentation on how to do kafka integration with citrine os.

masonseaman, 2025-07-14:
Websockets should work fine as long as citrineos-core is running, I would check to see if citrineos ran into issues starting up. 

Also, kafka support has been deprecated. We recommend RabbitMQ for best compatibility, due to its superior functionality in a production deployment.

masonseaman, 2025-07-15:
That should be the only place you need to make changes